fl4p/fugu-mppt-firmware #2: I'm also building this MPPT

augonis, 2023-05-31:
Hi, 

I'm also am developing MPPT based on this project. 
I've already designed my own PCB incorporating some of the hardware changes You mention in your notes (like low side shunt current sense and low side backflow MOSFET). Also have supplies figured out, but will have to verify the results.

Now I'm looking to tackle the firmware changes and found your fork, and I'll be building upon your fork.

Just letting you know, maybe we can de-duplicate our efforts.

fl4p, 2023-07-07:
Hi @augonis, I just saw your post.
Great, let's connect then. What EDA Software do you use? Can you share your schematics?

augonis, 2023-07-11:
Hi, 
I'm using EasyEDA, for it's free and integrates lcsc.com jlcpcb.com, which I used for PCB manufacturing and assembly (had 20 boards made). Also I stripped everything not needed for operation (LCD, USB serial converter, etc.).
I could share the schematic.

Firmware wise. I looked at your code and found myself not too familiar/comfortable with your code style (in the overall structure sense), so I decided to better start fresh project integrating some of your code where it made sense to me. 
I've since got a running firmware and I've put a good 10-20kWh through the two boards I prepared. Though the actual power point tracking seeps dodgy.

fl4p, 2023-07-12:
Thanks for your feedback and glad to hear you can use parts of my code. My background is more on the software side, this is my first bigger firmware project. I tried to logically structure things and I found this sometimes to be not ideal on a MCU with limited resources. Happy about any more input. Also like to see your code.

Have two boards running here, too. They each produce >4kWh per day, using the original PCB, with the hall sensor at the input of the half-bridge. I tried many ways to deal with the heavy noise; using a low tracker frequency (1-2 Hz) and ignoring any power change <1.5 W works best so far. Recording U,I,P, duty cycle and good visualisations helped a lot while tuning the algorithm.

A periodic global MPP scan every 15min deals with partially shaded strings and reveals local maximum, although I found that one of my 2S solar strings doesn't really need it.

We are planning to offer this as a commercial product, maybe as an assembly kit. Should have some educational value.
In case you want to connect, you find my email in my profile.

augonis, 2023-07-14:
I still need to clean up some stuff, but I hope I'll manage to publish my project next week.

augonis, 2023-07-19:
I've published my project, you can check it out https://github.com/augonis/pogo-MPPT

fl4p, 2023-07-29:
Cool, thanks for sharing.

I am also designing the new PCB with a INA chip. The INA228 works up to 48V and has an additional Vbus input, which can be multiplexed between Vin/Vout, so I can omit the ADS1015. There are also some cheaper INA chips with lower CMRR Voltage and Vbus.

Did you try the current sense with the ADS1015?

Here are my thoughts on your schematics:
To reduce ripple current in C7, you should place one or more ceramic caps of a few uF in parallel. I've noticed that this cap can produce a lot of heat. Placing the extra low ESR caps will increase conversion efficiency and lifetime of the electrolytic.
You might also place a RC Snubber from ground to switch node, to reduce switching stress and EMI.

I've re-structured my firmware here, to make it more readable.

augonis, 2023-07-29:
The INA219 in in the schematic so that I had an unpopulated footprint on the PCB in case I had to upgrade current sensor. In fact I am using the ADS1015 for current sense. The resolution isn't great, but it seems to work good enough.
INA228 is expensive, if better current resolution is needed I'd try an op-amp to to boost the shunt voltage first.

The buck part of the circuit I copied one-to-one from Angelos design. If I'll need to make more boards, I'll try to integrate your suggestions.

Just curious: Did you have a look on the firmware? Any feedback, Ideas to burrow?

Eventually I will write a Home Assistant integration to act as a hub for telemetry, settings, etc.

fl4p, 2023-08-01:
* Place header files in the source directory. This way it is less fragmented and easier to browse through the files especially when you're new to the code. This might be just personal taste
* I like the idea of using the RGB LED WS2812B. I might copy that
* I see you sample Vin, Vout, Iout consecutively and then notifying the buck control loop. This adds extra latency to your control loop. Better to use the samples as soon as they are available. Not sure how crucial this is in your application. I've noticed some +40% output voltage transients when the load is cut, which might destroy something still connected to the output. So you want your output voltage regulation to be as fast as possible. A quick fix would be to read Vout after Vin and Iout, so it is fresh when you notify control loop.

I see the buck code is mostly by me. Glad you could use it.

fl4p, 2023-08-02:
@augonis The datasheet of the WS2812B says it needs 3.7V supply and 3.5V logic level. I see you're running this on just the 3.3V and it works?

augonis, 2023-08-02:
I've used WS2812B on several projects on 3.3V and it works just fine. I like using it for indicators as it's a single component solution with just 3 wires to run/solder.

fl4p, 2023-08-31:
I've been struggling with the ADC noise at higher power levels (>600W). ADC readings have outliers and there was even interference on the UART. I tracked it down to be a ringing switch node at frequencies 10-30MHz (switch node to GND).

This was due to the high Qrr of the CSD19505, proposed in the original Fugu design. These fets are not suitable for high-side switching, rather synchronous rectification as the datasheet says. I replaced the CSD19505 with a STP150N10F7 and it works much better. Very little ringing and no noise issues. yay 

@augonis Did you check the voltage spikes at the switch node with an oscilloscope?
Looks like your crss035n10n has the same issue (Qrr=460).  Note that its datasheet doesn't mention DCDC converter in the application list. Look for a replacement with a Qrr < 200nC. There appears to be no manufacturer which includes the Qrr in the parametric search, so we have to look through the data-sheets.

* https://efficiencywins.nexperia.com/efficient-products/is-todays-accepted-MOSFET-figure-of-merit-still-relevant.html

augonis, 2023-09-01:
For now I haven't got a setup for higher power testing (likely won't have till next spring), but I did check just now at <200W (1P to 27V) and saw no spikes.

Haven't had much time to work on the MPPT (looks good enough for now), am trying to focus on installing the solar panels now.

Thanks for the info, I'll know what to watch out for when getting into higher power.