fl4p/fugu-mppt-firmware #3: Loop latency too high! shutdown

f4soft, 2023-09-02:
I have a problem. The output is turned off when "Loop latency too high! shutdown". I am using esp32 (original PCB) and added to the code in PinConfigESP32

Backflow_SD = 0,
INA22x_ALERT = 0,

because it was not compiled under esp32. All other code remained unchanged. Can you tell me why the "Loop latency too high! shutdown" problem occurs? The output is loaded only on a multimeter, a 24v power supply is connected to the input.

LOG:
[650164][E][main.cpp:300] loop(): [main] Loop latency too high! shutdown
Vi/o=25.83/ 9.65 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)=   0|   0| 123 MPPT(st= MPPT,0) lag=1547.8ms N=1060
PWM disabled (duty cycle was 101)
[650180][I][mppt.h:303] startSweep(): [mppt] Start sweep
[650180][I][sampling.h:188] startCalibration(): [mppt] Start calibration
[651648][I][sampling.h:257] update(): [sampler] Sensor I_raw calibration: avg=-0.2091 std=0.000000
[651649][I][sampling.h:259] update(): [sampler] Sensor I_raw midpoint-calibrated: -0.209061
[651657][I][sampling.h:257] update(): [sampler] Sensor U_out_raw calibration: avg=8.9315 std=0.000106
[651665][I][sampling.h:257] update(): [sampler] Sensor U_in_raw calibration: avg=25.8547 std=0.000001
[651672][I][sampling.h:266] update(): [sampler] Calibration done!
Vi/o=25.87/ 4.51 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C  0sps 10kbps PWM(H|L|Lm)=  88| 123| 123 MPPT(st=SWEEP,1) lag=1547.8ms N=118
[654540][I][mppt.h:323] _stopSweep(): [mppt] Stop sweep at controlMode=CV PWM=141, MPP=(0.0W,PWM=0,0.0V)
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 135| 123| 123 MPPT(st=   CV,1) lag=1547.8ms N=236
[656442][I][charger.h:39] getToppingCurrent(): [chg] Begin topping mode Iout_max 0.50 (start=14.02 V, release=13.72 V)
Vi/o=25.83/14.46 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 129| 123| 123 MPPT(st=   CV,1) lag=1547.8ms N=354
Vi/o=25.83/14.31 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 38sps 10kbps PWM(H|L|Lm)= 128| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=471
Vi/o=25.87/14.46 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 129| 123| 123 MPPT(st=   CV,1) lag=1547.8ms N=589
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 133| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=707
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 38sps 10kbps PWM(H|L|Lm)= 128| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=824
Vi/o=25.83/14.26 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 133| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=942
[677180][E][main.cpp:300] loop(): [main] Loop latency too high! shutdown
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 38sps 10kbps PWM(H|L|Lm)=   0|   0| 123 MPPT(st= MPPT,0) lag=1547.8ms N=1059
PWM disabled (duty cycle was 130)
[677195][I][mppt.h:303] startSweep(): [mppt] Start sweep
[677196][I][sampling.h:188] startCalibration(): [mppt] Start calibration
[678666][I][sampling.h:257] update(): [sampler] Sensor U_in_raw calibration: avg=25.8527 std=0.000000
[678673][I][sampling.h:257] update(): [sampler] Sensor I_raw calibration: avg=-0.2091 std=0.000000
[678674][I][sampling.h:259] update(): [sampler] Sensor I_raw midpoint-calibrated: -0.209061
[678682][I][sampling.h:257] update(): [sampler] Sensor U_out_raw calibration: avg=13.3691 std=0.000009
[678690][I][sampling.h:266] update(): [sampler] Calibration done!
[678696][I][charger.h:42] getToppingCurrent(): [chg] End topping mode, Vout 13.37
Vi/o=25.83/ 4.07 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C  0sps 10kbps PWM(H|L|Lm)=  87| 123| 123 MPPT(st=SWEEP,1) lag=1547.8ms N=117
[681498][I][mppt.h:323] _stopSweep(): [mppt] Stop sweep at controlMode=CV PWM=138, MPP=(0.0W,PWM=0,0.0V)
Vi/o=25.83/14.31 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 133| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=235
[683476][I][charger.h:39] getToppingCurrent(): [chg] Begin topping mode Iout_max 0.50 (start=14.02 V, release=13.72 V)
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 135| 123| 123 MPPT(st=   CV,1) lag=1547.8ms N=353
Vi/o=25.87/14.36 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 131| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=471
Vi/o=25.87/14.26 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 132| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=589
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 39sps 10kbps PWM(H|L|Lm)= 130| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=707
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 23°C 38sps 10kbps PWM(H|L|Lm)= 128| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=824
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 132| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=942
[704218][E][main.cpp:300] loop(): [main] Loop latency too high! shutdown
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 38sps 10kbps PWM(H|L|Lm)=   0|   0| 123 MPPT(st=   CV,0) lag=1547.8ms N=1059
PWM disabled (duty cycle was 132)
[704233][I][mppt.h:303] startSweep(): [mppt] Start sweep
[704234][I][sampling.h:188] startCalibration(): [mppt] Start calibration
[705703][I][sampling.h:257] update(): [sampler] Sensor U_in_raw calibration: avg=25.8543 std=0.000001
[705710][I][sampling.h:257] update(): [sampler] Sensor I_raw calibration: avg=-0.2091 std=0.000000
[705711][I][sampling.h:259] update(): [sampler] Sensor I_raw midpoint-calibrated: -0.209061
[705720][I][sampling.h:257] update(): [sampler] Sensor U_out_raw calibration: avg=13.3725 std=0.000009
[705727][I][sampling.h:266] update(): [sampler] Calibration done!
[705733][I][charger.h:42] getToppingCurrent(): [chg] End topping mode, Vout 13.37
Vi/o=25.87/ 4.07 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C  0sps 10kbps PWM(H|L|Lm)=  87| 123| 123 MPPT(st=SWEEP,1) lag=1547.8ms N=117
[708536][I][mppt.h:323] _stopSweep(): [mppt] Stop sweep at controlMode=CV PWM=138, MPP=(0.0W,PWM=0,0.0V)
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 131| 123| 123 MPPT(st=   CV,1) lag=1547.8ms N=235
[710389][I][charger.h:39] getToppingCurrent(): [chg] Begin topping mode Iout_max 0.50 (start=14.02 V, release=13.72 V)
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 131| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=353
Vi/o=25.83/14.36 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 38sps 10kbps PWM(H|L|Lm)= 131| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=470
Vi/o=25.83/14.31 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 132| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=588
Vi/o=25.83/14.36 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 130| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=706
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 131| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=824
Vi/o=25.87/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 39sps 10kbps PWM(H|L|Lm)= 130| 123| 123 MPPT(st= MPPT,1) lag=1547.8ms N=942
[731242][E][main.cpp:300] loop(): [main] Loop latency too high! shutdown
Vi/o=25.83/14.41 Ii/o= 0.0/ 0.0A Pin= -0.0W 24°C 38sps 10kbps PWM(H|L|Lm)=   0|   0| 123 MPPT(st= MPPT,0) lag=1547.8ms N=1059
PWM disabled (duty cycle was 133)
[731256][I][mppt.h:303] startSweep(): [mppt] Start sweep
[731257][I][sampling.h:188] startCalibration(): [mppt] Start calibration
[732727][I][sampling.h:257] update(): [sampler] Sensor U_out_raw calibration: avg=13.4214 std=0.000007
[732735][I][sampling.h:257] update(): [sampler] Sensor U_in_raw calibration: avg=25.8485 std=0.000000
[732743][I][sampling.h:257] update(): [sampler] Sensor I_raw calibration: avg=-0.2091 std=0.000000   
[732743][I][sampling.h:259] update(): [sampler] Sensor I_raw midpoint-calibrated: -0.209061
[732751][I][sampling.h:266] update(): [sampler] Calibration done!
[732757][I][charger.h:42] getToppingCurrent(): [chg] End topping mode, Vout 13.42

fl4p, 2023-09-03:
Hi,

I added this rate constrain to ensure that everything is OK with the ADC. 
I just pushed a fix which lowers the threshold.

you can also fix it yourself:
```
if (sps < 80 && !pwm.disabled() && nSamples > 1000 &&
```
replace 80 with 20

fl4p, 2023-09-03:
The 80 Hz works with the INA226. Sampling with the ADS1015 is a bit slower

f4soft, 2023-09-03:
Thank you. Now all work. I have INA226, but it has 36V max voltage(

f4soft, 2023-09-03:
Oh, I see I solder ADS1115 in future i replace it to ADS1015. Maybe better INA228 (up to 85V)? But I found about INA228 https://github.com/Zanduino/INA/issues/77#issuecomment-997338630. And i found new version from Angelo Casimiro with MOSFET HY4008  and HLSR 50-P https://www.facebook.com/photo/?fbid=4075260245853938&set=a.4075110759202220

fl4p, 2023-09-04:
> Thank you. Now all work. I have INA226, but it has 36V max voltage(

Ok good. I use the INA226 on the low-side ground current and a voltage divider for voltage sensing. For solar voltage is use the ESP32 ADC, it is not used for tracking, only diode emulation and OV/UV detection.
you can find the schematics on [cadlab](https://cadlab.io/project/27217/master/files)

fl4p, 2023-09-04:
> Oh, I see I solder ADS1115 in future i replace it to ADS1015. Maybe better INA228 (up to 85V)?

Yep ADS1015 is faster. Unless you need a Common Ground (Solar GND = Bat GND), the INA226 works fine. INA228 is expensive.

> But I found about INA228 [Zanduino/INA#77 (comment)](https://github.com/Zanduino/INA/issues/77#issuecomment-997338630).

Interesting project, I didn't know. I used code for the INA228 based on [craigpeacock/Linux_PowerMeter ](https://github.com/craigpeacock/Linux_PowerMeter)
for my [smart-shunt power monitor](https://github.com/open-pe/smart-shunt-fw)

fl4p, 2023-11-30:
@f4soft how is it going? any progress on your MPPT project?

closed 2024-11-29 by fl4p