fl4p/fugu-mppt-firmware #4: two mppt tracker with one esp32 possible

jbaltes, 2023-09-06:
Hi,

did you know whether there are enough CPU resources for two seperated mppt tracker running in one esp32?

best regards
Jochen

fl4p, 2023-09-07:
Hi Jochen,
yes should be feasible. Current update rate is about 200 Hz without telemetry.
Adding more channels will lower the rate, trading in for V_out control speed.
I'll add this to the todo list.

Also want to do cpu profiling, to find out the critical parts. Maybe its worth switching from float to integer arithmetic

SCiunczyk, 2024-08-04:
Hi,
I'm I'm very interested with even more...
On my yacht I have four solar panels installed on deck. Due to yacht nature heel & pitch & roll and lot of obstacles shading different parts of random panels each panel is working on his very own PP. I think that ideal solution would be four separate MPPT controlers with outputs connected in parallel. Maybe increasing clock of esp32 to 240MHz and changing math to integer will give enough power to have one esp32 with four MPPT? 
I'm ready to jump into that project.

ps
Asked on oryginal instructables but looks that project is not very active