fl4p/fugu-mppt-firmware #7: PCB reference

Warm-rain, 2023-12-13:
I have many open source PCB circuit diagrams here, which may be helpful for your design.
https://oshwhub.com/q12888873/mppt9635a
https://oshwhub.com/muyan2020/wei-yan-zheng-esp32-guang-fu-pv-mppt-ina226_copy_copy_copy_copy_copy

fl4p, 2023-12-13:
Thank you for sharing.
Interesting to see a buck-boost topology . Have you tried to use it with a wind generator?

I see you use the CSD19505KCS for high side (Q5, Q13). Did you measure the voltage wave form at switching nodes? 
These switches are not designed for HS switching due to high Qrr and can cause a lot of ringing and EMI (also see my https://github.com/fl4p/fugu-mppt-firmware/issues/2#issuecomment-1701801832) 

Maybe you are interested in seeing my PCB design here: https://cadlab.io/project/27217/master/files

Warm-rain, 2023-12-14:
You can try these measures, but unfortunately they can only be alleviated, not completely eliminated.
[img]https://i.imgs.ovh/2023/12/14/mt0Dt.jpeg[/img]