fl4p/fugu-mppt-firmware #14: Support for Sodium Ion?

samuk, 2024-11-05:
Given increasing availability of Sodium-ION batteries https://vi.aliexpress.com/w/wholesale-sodium-ion.html

Would the Fugu2 board support this chemistry?

fl4p, 2024-11-07:
I have no experience with this kind of batteries. From what I got from a quick google, is that they can be charged similar to LFP (constant-current, constant-voltage).
If this is true, its should work straight away. 

You just need to adjust battery OCV at 100% in
https://github.com/fl4p/fugu-mppt-firmware/blob/d698e3350bbf96cabd58ca4f6d5db06b02c8c4e5/src/battery.h#L23

Do you know more about this chemistry? Maybe you have a data sheet?

https://www.batterydesign.net/sodium-ion-battery/

samuk, 2024-11-07:
Nice. I don't know much about them either as they are pretty new on the market. There don't seem to be many charging solutions for them, so good to see it should be fairly straightforward with your MPPT

closed 2024-11-07 by samuk