fl4p/fugu-mppt-firmware #15: Fugu2 hardware: USB-C?

samuk, 2024-11-05:
I was wondering if you'd consider doing a version of your Fugu2 hardware that included a USB-C & USB-A socket?

The PD stuff could be stolen from https://oshwlab.com/wagiminator/ip6520-usb-pd-source

Or alternately maybe spin it out as a daughter board?

samuk, 2024-11-06:
Wrong repo

closed 2024-11-06 by samuk

fl4p, 2024-11-07:
yes, here's the daughter board with usb type-c connector: https://github.com/open-pe/mcu-head-p12-esp32s3-wroom

it can be mounted with 2.54 SMD pin headers

fl4p, 2024-11-08:
This little board can be placed on the Fugu2 board instead of the ESP32-S3-WROOM, using SMD pin header.

I currently don't have a rendering or photo to show. Here you can see the footprint of the mounting header to be placed instead of the WROOM:

[image]

fl4p, 2024-12-13:
Hi, i was finally able to build the new design
[image]

fl4p, 2024-12-13:
as you can see the esp32-wroom is mounted on a little shield, which has an usb type c plug (not visible on the picture)

i might change the interface for this, so it's compatible with an Esp23s3-zero

[image]