fl4p/fugu-mppt-firmware #19: Writing to Influxdb

tertomium, 2024-11-20:
Hey, trying to get Influxdb 2 to work but struggling to do so.
In telemetry.h I set my host (server) IP and also the port my Udp listener is configured to (8089).
I’ve configured the listener with it’s needed Token and also set a bucket. I don’t get any data….sending Udp packet manually creates a data entry in influxdb2 so this part seems to work. 
Is this feature already working or still in development?

Regards 

Steffen

fl4p, 2024-11-22:
Hi Steffen,
this was previously working, until I added a buffer queue for points to be sent to lower control loop latency.
Reading the queue is currently not implemented, see`telemetryFlushPointsQ()`.

You could easily implement it by reading some Points into a local array and passing it to `influxWritePointsUDP()`.
This would add additional overhead and it would be better to consume from the queue until MTU (max UDP packet size) is reached.

I'll try to implement that and let you know.

tertomium, 2024-11-22:
Hey, thanks for answering me. I’m still running the 2.1.0 Version from last year i think. 
In Telegraf debug I can see it writes metrics but nothing show up in the bucket. Maybe a timestamp problem? Do u use influx data format and Linux timestamp?

fl4p, 2024-11-22:
It uses unix timestamps. Did you check the serial console for time sync? I think it needs to have a NTP sync for proper timestamping.
Maybe you find points dated at the year 1970 in influxdb?

tertomium, 2024-11-22:
Time on display after boot is correct so this should work. I already tried dating back but still no  luck. Bet it’s something simple. Sending packets with PacketSender works and data shows up in my bucket.

tertomium, 2024-11-23:
Got it. I had to switch to nanoseconds _"point.setTime(WritePrecision::NS)"_ in telemetry.h and mppt.h.
Now data is dropping in as it should.