fl4p/fugu-mppt-firmware #20: The output current is twice as high.

tertomium, 2024-11-24:
Im using the mppt currently with a 12V/8.4Ah AGM Battery.
I set FUGU_BAT_V 14.4 ( Cyclic Use Voltage is 14.4V-15.0V) in "charger.h" and in battery.h i set:

`float detectMaxBatteryVoltage(float idleVoltage) {
    float voltageStep = 14.4f; // lifepo4 4s (3.65V) [1,2,4]
    float minVoltage = 10.f;`

Now while using a variable power supply i get twice the output current. I have to set in main.cpp:
`sensors.Iout = (Iout_ch != 255)
                   ? adcSampler.addSensor(
                    {
                            Iout_ch,
                            Iout_transform,
                            {2.0f, .1f, true},
                            "Io",
                            true},
                    30.f, ioutFiltLen)
                   : adcSampler.addVirtualSensor([&]() {
                if (std::abs(sensors.Iin->last) < .05f or sensors.Vout->last < 0.1f)
                    return 0.f;
                return ((sensors.Iin->last * sensors.Vin->last * conversionEfficiency) / (sensors.Vout->last)/2);
            }, ioutFiltLen);`

Dividing the return by 2 for the virtual sensor.

Obviously im missing something here... Since youre using 28V Battery System maybe its related to this?

fl4p, 2024-11-25:
Which commit hash are you working with? `git rev-parse --short HEAD`

Is `conversionEfficiency` a local variable? If so, replace `adcSampler.addVirtualSensor([&]()` with `adcSampler.addVirtualSensor([]()`.

I remember there was a bug with local variable `conversionEfficiency` being referenced by the virtual sensor function, which caused access to an invalid stack address.

tertomium, 2024-11-25:
I will check when I’m back home. I’m On-Site for a couple of days. I’ve installed everything to my wind turbine and it did work. You can hear it through the generator when tracking is ongoing ;-)
I need to reinforce my mast or install some tie-downs first. I really underestimated the forces…