fl4p/fugu-mppt-firmware #31: Implementing Cutoff current logic and CANBus comms feature

shvmm, 2025-04-01:
## Cutoff (AKA V minus I×R) Charging logic

Cutoff charging logic deviates from the routine CC-CV phase in that we stop charging when
[image]

For LFP battery chemistries, CVmin is taken to be 3.37V and CVmax as 3.65V

## CANBus
Comms feature can be presently implemented on the fugu2 PCB space devoting a header for the popular SN65HVD230 breakout board.

[image]

## CAN protocol

For the aforementioned cutoff charging logic,
Battery Ah, Charging current and voltage needs to be recognised.
These are sent by CAN IDs 0x351 and 0x359 from the BMS :
### CAN ID 0x351
| Byte | Definition | Unit| Note |
|--------|--------|--------|--------|
| Byte 0 to 1 | CVL | 0.1V | Requested Charging Voltage LIMIT |
| Byte 2 to 3 | CCL | 0.1A | Requested Charging Current LIMIT |

### CAN ID 0x359
| Byte | Definition | Unit| Note |
|--------|--------|--------|--------|
| Byte 5 to 6 | Capacity | 1Ah | Battery capacity in Ah |

frame 500kbps, little endian, data is expected to receive every 1s
MPPT ACK response: CAN ID: 0x305: Data: 00 00 00 00 00 00 00 00

fl4p, 2025-10-29:
Thanks @shvmm , great points.

I was not aware of this LFP charging logic. I'll try to implement it when I have time.

CANbus is on my list too. Maybe I'll add the SN65HVD230 directly to the fugu2 pcb. UART is generally not needed anymore for programming, since the JTAG USB is working fine on the ESP32-S3. I've removed the Pin-Header from the latest board and instead put a USB-C socket and UART testpoints for solder connections. 

The devices I use currently communicate completely wirelessly over WiFI, which works well so far.

fl4p, 2025-11-12:
> In a broadcast configuration, all listening devices will receive the same message at the same time.

Yeah, we need to specify some communication protocol between the leader device and the followers. My idea is: The followers need to register with the leader, so the leader knows how many devices there are and then either broadcasts the same to all followers or talk to each one individually. Maybe we find a standard that already implements something similar, although it shouldn't be that hard to program. It of course requires that the CAN bus doesn't get too busy and there are no conflicting CAN IDs (the standard format has 11 bit only, which means 2047 unique IDs).
Software-wise it is more complex than daisy-chaining, but it needs less hardware, which I'd prefer.

shvmm, 2025-12-25:
Original motivation was after reading this article https://nordkyndesign.com/charging-marine-lithium-battery-banks
And realising most LFP cell datasheets echo the same thing such as these:

[CALB-L160F100.pdf](https://github.com/user-attachments/files/24335639/CALB-L160F100.pdf)

It is just simple V-IR based linear regression to adapt charging behaviour for non-constant current real world charging conditions.

Here is a description of the LFP charging flow.

[image]

shvmm, 2025-12-25:
> The charger tracks average battery current and its own output current. With that we compute the residual current Irest = Iout-Ibat (from other chargers and loads), which will be the target output current once the termination criteria becomes true.

In BMS comms mode, we already have the advantage that the amps communicated by the BMS will always be the current going/in out of the cells.

We can directly use this current value I_bat as follows:

```
// CPP sample Pseudocode
// cell_ah = user supplied, 100Ah as example
// CVmin = 3.375, CVmax = 3.625, Cutoff_Rate (k) = 0.05 C (k can be increased to 0.08 C or even 0.1C for extra safety margin)
                // cutoff_resistance: (CVmax - CVmin) ÷ (k * Ah)
                // cutoff_r = (3.625 - 3.375) ÷ (0.05 * Ah) = 5 / Ah
                float cutoff_r = (3.625-3.375) / cell_ah / 0.05;

static num_count = 0;
void charging_loop (args... if any) {
    
    if (max_cell_voltage - cutoff_r * I_bat) > CVmin && I_bat > 0) {
        // Current can't be negative for calculations.
        n++;
        // the criteria check must succeed n=4 times in a row in order to safely terminate,
        //so we don't terminate with random/bogus/erroneous readings from BMS
    } else n = 0; // resets n back to 0 otherwise
    
    if (n>4) {  
        // charge termination is reached.
        // The highest voltage cell is fully charged, charging must be terminated and we can proceed with balancing.
        // Do something here
     }
}
```

shvmm, 2025-12-26:
>> AutoCVL

> might take this name.

With Pleasure. AutoCVL/CCL stands for Auto(matic) Charging Voltage/Current Limiting

>Do you know under which names the V-I*R logic is referred to? Still looking for a suitable name

My bad. It is just another way to express the cutoff-charging logic. There is no difference, they're identical.

> Maybe C is missing in your equation @shvmm ? Or that k := 0.05 C

Maybe I forgot but it was the very first post of this discussion: (Ah is cell_capacity)

> ## Cutoff (AKA V minus I×R) Charging logic
> Cutoff charging logic deviates from the routine CC-CV phase in that we stop charging when [image]

> Now after further thinking about it, it probably makes sense to merge both approaches, and start reducing when highest cell reaches CVmin, and ramp the current down to zero until CVmax is reached.

The end-user can provide any charging voltage limit in the range [CVmin, CVmax] = [3.375V, 3.625V].
Suppose, the user provides 3.425V, we CC-CV upto 3.425V max_cell_voltage and allow AutoCVL to hold the cell voltage until cell_current starts to taper down and cutoff criteria is met at 0.01C.

If a user provides 3.55V, we CC-CV upto 3.55V max_cell_voltage and allow AutoCVL to hold that voltage until cell_current starts to taper down and cutoff criteria is met at 0.035C.

> As compared to a hard binary cut-off, I believe this allows us to push more charge into the battery. (A "normal" BMS will always cut-off at 3.65V, although we can go beyond the 3.65V if current is sufficiently high without overcharge)

Under no circumstance, cell_voltage should be allowed to increase beyond 3.65V - some safety margin (0.02V?)

> I am still not sure wether its useful to have the charger go to a "charge terminated" state that is released once voltage drops back to CVmin.

Happy to clarify this misunderstanding. For the charger to attempt another charge cycle, the trigger cell_voltage must always be sufficiently lower than CVmin, like around 3.32-3.28V or so. If you want to prevent frequent ReBulk triggering due to cell_voltage dip from momentary loads, V-IR can be used even for ReBulk criteria.

Any voltage in the range [CVmin, CVmax] is valid (charging voltage limit) that can fully charge a LFP cell provided time. All the Cutoff-logic does is detecting when a cell is fully charged and terminate charging.
The full charge voltage for LFP is around ~3.37 with no current flowing into the cell.

fl4p, 2026-01-03:
> My bad. It is just another way to express the cutoff-charging logic. There is no difference, they're identical.

I see, I thought with AutoCVL you were referring to something else

> (Ah is cell_capacity)

ok, then its the same equation.

After some experimenting with the batteries and reading about the topic my latest implementation works like this:
It now computes the termination cell voltage:

$$V_{term} = CVmin + I \cdot \frac{CVmax - CVmin}{k\cdot C} $$

Once the highest cell reaches this voltage, the charger limits pack voltage to the current output voltage (CV phase). If the communication to the BMS is interrupted, $V_{term}$ falls back to $CVmin$.

I removed the current fade-off mechanism I mentioned in a previous comment, as this doesn't appear to be useful. During CC phase (bulk) it just pumps everything the solar panels provide in the battery, until termination condition and turns into CV mode ("absorption"). In CV phase battery current will fall, further reducing termination voltage, which quickly leads to zero battery current. Resting above CVmin is considered to be unhealthy.

Implementation might change again, after further testing with batteries (I use a 8s 280Ah prismatic cells aliexpress pack)

fl4p, 2026-01-11:
> An end user can supply any cell_voltage in range [3.37, ~3.65] as their preference for charging voltage. Most people use something around ~3.5V/Cell

With "charging voltage", you referring to CVmax?

CapnKernel, 2026-01-12:
> There is no reason not to charge fully a LFP cell everytime to 100%.

LFP rocks eh!

Does Fugu2 support other lithium chemistries, such as NMC?