fl4p/fugu-mppt-firmware #32: Is there an error on the ACS712?

noeldum, 2025-04-02:
See the diagram for the ACS712.

[image]

It looks like the pin to be connected to the 2.2uf capacitor is FILTER and not VOUT.

[image]

The datasheet also suggest this.

fl4p, 2025-04-03:
Hi,

The first circuit you posted does make little sense.

[image]

VIOUT is a low impedance output and directly connecting a capacitor to GND will not apply a useful filtering.

noeldum, 2025-04-03:
sorry I posted in the wrong repo. I'll delete the post.

closed 2025-04-03 by noeldum